v4l2loopback/v4l2loopback #561: Chromium cannot read camera: Dequeued v4l2 buffer contains invalid length

hackermondev, 2023-10-24:
I installed the v4l2loopback kernel module on my machine and ran it with ``sudo modprove v4l2loopback exclusive_caps=1``. 

I created a camera on ``/dev/video0`` with [the rust bindings](https://docs.rs/v4l2loopback/latest/v4l2loopback/) and starting piping a static image to the camera (command from the wiki):
```bash
sudo ffmpeg -loop 1 -re -i 60828015.jpg -f v4l2 -vcodec rawvideo -pix_fmt yuv420p /dev/video0
```

I'm running this on a VPS so I quickly spun up a Docker container with a VNC and desktop environment to test everything (``sudo docker run --rm -it --device /dev/video0 --privileged -p 8090:8080 theasp/novnc``). Everything looks fine and if i run ``ffplay /dev/video0``, I can successfully view the image in the camera. The problem now is with other applications

Cheese doesn't even detect the camera. Chromium detects the camera but cannot use it:
```
[15749:15755:1024/025614.520951:ERROR:v4l2_capture_delegate.cc(1138)] Dequeued v4l2 buffer contains invalid length (11441 bytes).
```

Not sure what I'm doing wrong and why Chromium cannot read from the camera.
- kernel version: 5.4.0-164-generic
- v4l2loopback version: commit 5bb9bed on the main branch (latest one right now)
- module paramaters: exclusive_caps=1

hackermondev, 2023-10-27:
@umlaeute Any chance you know what's going on here?

hackermondev, 2023-10-28:
The issue was with the size of the image I was sending with ffmpeg (87x87). I'm assuming there's a minimum webcam frame size in Chromium and I wasn't aware of it. I started sending a larger image with ffmpeg (2880x1800) and it no longer logs that error and the webcam works.

closed 2023-10-28 by hackermondev

umlaeute, 2023-10-29:
Depending on the colour encoding, odd-sized frames can be problematic (eg YUV420p requires that the U and V planes are downsampled by a factor of 2, which works best if the width and height can be divided by 2)

hackermondev, 2023-10-31:
@umlaeute Yep, looks like that was the issue. Maybe add that to the wiki?

umlaeute, 2023-10-31:
Sure, please go ahead!