v4l2loopback/v4l2loopback #562: DKMS Won't Sign When Building

v4u6h4n, 2023-11-08:
v4l2loopback: v0.12.7
Kernel: v6.1.0-13-amd64
Distro: Debian 12

---

Hey everyone :-) I didn't have any luck with DKMS signing v4l2loopback-dkms for my secure boot when installing with apt, so I tried [the instructions](https://github.com/umlaeute/v4l2loopback#dkms) in readme.md to manually build with DKMS, however I encountered the same error when doing so. Any advice would be appreciated.

> At main.c:170:
> 
> SSL error:07880109:common libcrypto routines::interrupted or cancelled: ../crypto/passphrase.c:184
> SSL error:07880109:common libcrypto routines::interrupted or cancelled: ../crypto/passphrase.c:184
> SSL error:1C80009F:Provider routines::unable to get passphrase: ../providers/implementations/encode_decode/decode_epki2pki.c:96
> SSL error:07880109:common libcrypto routines::interrupted or cancelled: ../crypto/passphrase.c:184
> SSL error:04800068:PEM routines::bad password read: ../crypto/pem/pem_pkey.c:159
> sign-file: /var/lib/shim-signed/mok/MOK.priv
> Done.

umlaeute, 2023-11-08:
If DKMS doesn't sign, this is likely to be an issue with DKMS rather than with `v4l2loopback`.

Since DKMS is an external tool (outside the scope of this repository), I'm closing this as "not-for-us"

closed 2023-11-08 by umlaeute

v4u6h4n, 2023-11-09:
Alright I made a post in the DKMS github, will post here again with an update if theres anything worth updating in the readme.md

umlaeute, 2023-11-09:
README updates would be very much welcome, thanks.