v4l2loopback/v4l2loopback #566: Dummy device made via v4l2loopback is busy

igorhoop, 2023-12-20:
Hi @umlaeute . 

I use **v4l2loopback** on my laptop and and your program worked fine.
But now I'm trying to run it on **embedded** computers and it doesn't work. For example, now I am trying to run on NVidia Jetson. I connected a USB webcam:

`sudo modprobe v4l2loopback video_nr=99`

Now l have 2 devices:

```
ls /dev/video*

video0      # real device
video99    # dummy device
```

Next, I direct the real stream to the virtual stream using **ffmpeg**:
`ffmpeg -f v4l2 -i /dev/video0 -f v4l2 /dev/video99`

Next, I run **VLC** and view the image from **video99**. At this step everything works.
Now I want to run a copy of VLC and view **video99** again. That is, at the same time 2 VLCs must show the image from **video99**

But I get a error:
`v4l2 stream error: **cannot start streaming: Device or resources busy**`

If I turn off the first VLC instance, then the second VLC instance starts working normally.

I don't understand why this is happening. Perhaps I did something wrong, but it worked on my laptop.

Information that is installed on NVidia Jetson:
```
OS: Ubuntu 18.04
Kernel: 4.9.140-tegra

```

umlaeute, 2023-12-21:
so which exact version of `v4l2loopback` are you running on both your laptop and the jetson?

igorhoop, 2023-12-22:
> so which exact version of `v4l2loopback` are you running on both your laptop and the jetson?

I used the current version. Main branch.

Downloaded like this:
`git clone https://github.com/umlaeute/v4l2loopback.git`

umlaeute, 2023-12-22:
That's not an exact version.

Could you please post the version as printed in the kernel logs when the module is loaded? (`dmesg | grep v4l2loopback`)

umlaeute, 2023-12-22:
Thanks. That looks like three different versions...

umlaeute, 2024-03-19:
as of https://github.com/umlaeute/v4l2loopback/issues/310, it is not allowed to have multiple consumers reading in parallel.

closed 2024-03-19 by umlaeute