v4l2loopback/v4l2loopback #567: wont compile on armv7l

maxi322, 2023-12-27:
**My environment:**

  * `v4l2loopback` version: 850a2e36849f6ad3c9bf74f2ae3f603452bd8a71

  * kernel version: **Linux campi 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux**
  also tested on the latest rasberry pi kernel 6.1

  * Distribution (+version): Tested on Debian 10 and 12

### Step 3: Describe the problem:
The kernel module wont compile on 32 bit raspberry pi os.

#### Steps to reproduce:

1. Clone the repository on a raspberry pi running 32 bit raspberry pi os.
2. Try to compile it.

It will fail with:
  ```
ERROR: modpost: "__aeabi_ldivmod" [/home/pi/v4l2loopback/v4l2loopback.ko] undefined!
  ```

umlaeute, 2023-12-28:
this is probably a duplicate of https://github.com/umlaeute/v4l2loopback/issues/550

maxi322, 2023-12-28:
Yes you are right. Looking at the mentioned issue it does not look like this will get solved soon.
I switched to an aarch64 kernel in the meantime and there it compiles without a problem.

It would still be nice if this gets resolved some day as this kernel module can now no longer be used on the raspberry pi 1 and 2.

umlaeute, 2023-12-28:
i agree. i386 (as of #550) is probably not a very relevant architecture anymore, but i guess 32bit arm is going to stay around for a bit longer

closed 2024-03-21 by umlaeute