v4l2loopback/v4l2loopback #574: Fails to build against Kernel 6.8-rc6

DarkXero-dev, 2024-02-26:
Hello..

I am currently on Arch with KDE Plasma 6.0 and trying out Kernel 6.8-rc6 and **v4l2loopback-dkms** fails to build against that Kernel.. 

First Message
```
==> dkms install --no-depmod v4l2loopback/0.12.7 -k 6.8.0-rc6-273-tkg-eevdf
Error! Bad return status for module build on kernel: 6.8.0-rc6-273-tkg-eevdf (x86_64)
Consult /var/lib/dkms/v4l2loopback/0.12.7/build/make.log for more information.
==> WARNING: `dkms install --no-depmod v4l2loopback/0.12.7 -k 6.8.0-rc6-273-tkg-eevdf' exited 10
```

make.log message
```
DKMS make.log for v4l2loopback-0.12.7 for kernel 6.8.0-rc6-273-tkg-eevdf (x86_64)
   2   │ Mon Feb 26 09:03:42 PM EET 2024
   3   │ Building v4l2-loopback driver...
   4   │ make -C /usr/lib/modules/6.8.0-rc6-273-tkg-eevdf/build M=/var/lib/dkms/v4l2loopback/0.12.7/build modu
       │ les
   5   │ make[1]: Entering directory '/usr/lib/modules/6.8.0-rc6-273-tkg-eevdf/build'
   6   │   CC [M]  /var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.o
   7   │ /var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.c: In function ‘vidioc_querycap’:
   8   │ /var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.c:717:9: error: implicit declaration of function
       │  ‘strlcpy’; did you mean ‘strscpy’? [-Werror=implicit-function-declaration]
   9   │   717 |         strlcpy(cap->driver, "v4l2 loopback", sizeof(cap->driver));
  10   │       |         ^~~~~~~
  11   │       |         strscpy
  12   │ cc1: some warnings being treated as errors
  13   │ make[3]: *** [scripts/Makefile.build:243: /var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.o] Err
       │ or 1
  14   │ make[2]: *** [/usr/lib/modules/6.8.0-rc6-273-tkg-eevdf/build/Makefile:1924: /var/lib/dkms/v4l2loopbac
       │ k/0.12.7/build] Error 2
  15   │ make[1]: *** [Makefile:240: __sub-make] Error 2
  16   │ make[1]: Leaving directory '/usr/lib/modules/6.8.0-rc6-273-tkg-eevdf/build'
  17   │ make: *** [Makefile:43: v4l2loopback.ko] Error 2
```

Any help appreciated.

closed 2024-02-26 by umlaeute

DarkXero-dev, 2024-02-26:
Should I do something?

umlaeute, 2024-02-26:
use `v4l2loopback` from git.