v4l2loopback/v4l2loopback #575: dkms build fails on kernel version >6.8.0

jschwender, 2024-03-12:
build fails with error on strlcpy() function.

gcc (Ubuntu 13.2.0-4ubuntu3) 13.2.0

  * `v4l2loopback` version: 0.12.7
  * Distribution (+version): _____
          lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 23.10
Release:	23.10
Codename:	mantic

#### Steps to reproduce:

build vanilla kernel 6.8.0
run dkms build on installation the kernel

#### Observed Results:
build fails

Suggested fix:
--- v4l2loopback.c.sav	2023-07-05 14:05:05.000000000 +0200
+++ v4l2loopback.c	2024-03-12 08:36:48.736536585 +0100
@@ -739,7 +739,11 @@
 			->devicenr;
 	__u32 capabilities = V4L2_CAP_STREAMING | V4L2_CAP_READWRITE;
 
+#if LINUX_VERSION_CODE < KERNEL_VERSION(6, 8, 0)
 	strlcpy(cap->driver, "v4l2 loopback", sizeof(cap->driver));
+#else
+	strscpy(cap->driver, "v4l2 loopback", sizeof(cap->driver));
+#endif
 	vidioc_fill_name(cap->card, sizeof(cap->card), devnr);
 	snprintf(cap->bus_info, sizeof(cap->bus_info),
 		 "platform:v4l2loopback-%03d", devnr);
@@ -1247,7 +1251,11 @@
 	if (!cnf)
 		BUG();
 
+#if LINUX_VERSION_CODE < KERNEL_VERSION(6, 8, 0)
 	strlcpy(q->name, cnf->name, sizeof(q->name));
+#else
+	strscpy(q->name, cnf->name, sizeof(q->name));
+#endif
 	q->default_value = cnf->def;
 	q->type = cnf->type;
 	q->minimum = cnf->min;
@@ -1352,7 +1360,11 @@
 	memset(outp, 0, sizeof(*outp));
 
 	outp->index = index;
+#if LINUX_VERSION_CODE < KERNEL_VERSION(6, 8, 0)
 	strlcpy(outp->name, "loopback in", sizeof(outp->name));
+#else
+	strscpy(outp->name, "loopback in", sizeof(outp->name));
+#endif
 	outp->type = V4L2_OUTPUT_TYPE_ANALOG;
 	outp->audioset = 0;
 	outp->modulator = 0;
@@ -1411,7 +1423,11 @@
 	memset(inp, 0, sizeof(*inp));
 
 	inp->index = index;
+#if LINUX_VERSION_CODE < KERNEL_VERSION(6, 8, 0)
 	strlcpy(inp->name, "loopback", sizeof(inp->name));
+#else
+	strscpy(inp->name, "loopback", sizeof(inp->name));
+#endif
 	inp->type = V4L2_INPUT_TYPE_CAMERA;
 	inp->audioset = 0;
 	inp->tuner = 0;

replacing the function for kernels > 6.8 works for me, but i don't feel involved enough with the code, so i want you to look over it and decide if that is a good idea or not. Hope it helps.

edmundlaugasson, 2024-03-18:
I can confirm with EndeavourOS:
```
DKMS make.log for v4l2loopback-0.12.7 for kernel 6.8.1-arch1-1 (x86_64)
Sunday, March 17 2024 10:33:34 EET
Building v4l2-loopback driver...
make -C /usr/lib/modules/6.8.1-arch1-1/build M=/var/lib/dkms/v4l2loopback/0.12.7/build modules
make[1]: Entering directory '/usr/lib/modules/6.8.1-arch1-1/build'
  CC [M]  /var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.o
/var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.c: In function ‘vidioc_querycap’:
/var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.c:717:9: error: implicit declaration of function ‘strlcpy’; did you mean ‘strscpy’? [-Werror=implicit-function-declaration]
  717 |         strlcpy(cap->driver, "v4l2 loopback", sizeof(cap->driver));
      |         ^~~~~~~
      |         strscpy
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:243: /var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.o] Error 1
make[2]: *** [/usr/lib/modules/6.8.1-arch1-1/build/Makefile:1921: /var/lib/dkms/v4l2loopback/0.12.7/build] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.8.1-arch1-1/build'
make: *** [Makefile:43: v4l2loopback.ko] Error 2

```

closed 2024-03-19 by umlaeute

command-z-z, 2024-03-21:
> I can confirm with EndeavourOS: kernel 6.8.1-arch1-1 v4l2loopback-dkms 0.12.7-2 v4l-utils 1.26.1-1
> 
> ```
> DKMS make.log for v4l2loopback-0.12.7 for kernel 6.8.1-arch1-1 (x86_64)
> Sunday, March 17 2024 10:33:34 EET
> Building v4l2-loopback driver...
> make -C /usr/lib/modules/6.8.1-arch1-1/build M=/var/lib/dkms/v4l2loopback/0.12.7/build modules
> make[1]: Entering directory '/usr/lib/modules/6.8.1-arch1-1/build'
>   CC [M]  /var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.o
> /var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.c: In function ‘vidioc_querycap’:
> /var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.c:717:9: error: implicit declaration of function ‘strlcpy’; did you mean ‘strscpy’? [-Werror=implicit-function-declaration]
>   717 |         strlcpy(cap->driver, "v4l2 loopback", sizeof(cap->driver));
>       |         ^~~~~~~
>       |         strscpy
> cc1: some warnings being treated as errors
> make[3]: *** [scripts/Makefile.build:243: /var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.o] Error 1
> make[2]: *** [/usr/lib/modules/6.8.1-arch1-1/build/Makefile:1921: /var/lib/dkms/v4l2loopback/0.12.7/build] Error 2
> make[1]: *** [Makefile:240: __sub-make] Error 2
> make[1]: Leaving directory '/usr/lib/modules/6.8.1-arch1-1/build'
> make: *** [Makefile:43: v4l2loopback.ko] Error 2
> ```

`v4l2loopback/0.13.1.r8.gc94a1e8` fix the problem, you can update this package:)

Abdull, 2024-07-22:
I now run into this issue with Debian 12 during `apt upgrade` since the release of the [latest linux-image in *bookworm-backports*, `linux-image-6.9.7+bpo-rt-amd64`](https://packages.debian.org/bookworm-backports/linux-image-6.9.7+bpo-rt-amd64).

The previous [`linux-image-6.7.12+bpo-amd64`](https://packages.debian.org/bookworm-backports/linux-image-6.7.12+bpo-amd64) did not have this issue (so Linux `6.7.12` is okay, Linux `6.9.7` makes problem &mdash; fitting *this* issue report's "*dkms build fails on kernel*").

----

<details>
<summary>`sudo apt upgrade` output, containing error log information (click to reveal)</summary>

```
 $ sudo apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  linux-headers-6.1.0-22-amd64 linux-headers-6.1.0-22-common linux-image-6.1.0-22-amd64
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
4 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n]
Setting up linux-image-6.9.7+bpo-rt-amd64 (6.9.7-1~bpo12+1) ...
/etc/kernel/postinst.d/dkms:
dkms: running auto installation service for kernel 6.9.7+bpo-rt-amd64.
Sign command: /lib/modules/6.9.7+bpo-rt-amd64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j12 KERNELRELEASE=6.9.7+bpo-rt-amd64 KERNEL_DIR=/lib/modules/6.9.7+bpo-rt-amd64/build all...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.9.7+bpo-rt-amd64 (x86_64)
Consult /var/lib/dkms/v4l2loopback/0.12.7/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
dkms: autoinstall for kernel: 6.9.7+bpo-rt-amd64 failed!
run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
dpkg: error processing package linux-image-6.9.7+bpo-rt-amd64 (--configure):
 installed linux-image-6.9.7+bpo-rt-amd64 package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of linux-image-rt-amd64:
 linux-image-rt-amd64 depends on linux-image-6.9.7+bpo-rt-amd64 (= 6.9.7-1~bpo12+1); however:
  Package linux-image-6.9.7+bpo-rt-amd64 is not configured yet.

dpkg: error processing package linux-image-rt-amd64 (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of linux-headers-6.9.7+bpo-rt-amd64:
 linux-headers-6.9.7+bpo-rt-amd64 depends on linux-image-6.9.7+bpo-rt-amd64 (= 6.9.7-1~bpo12+1) | linux-image-6.9.7+bpo-rt-amd64-unsigned (= 6.9.7-1~bpo12+1); however:
  Package linux-image-6.9.7+bpo-rt-amd64 is not configured yet.
  Package linux-image-6.9.7+bpo-rt-amd64-unsigned is not installed.

dpkg: error processing package linux-headers-6.9.7+bpo-rt-amd64 (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of linux-headers-rt-amd64:
 linux-headers-rt-amd64 depends on linux-headers-6.9.7+bpo-rt-amd64 (= 6.9.7-1~bpo12+1); however:
  Package linux-headers-6.9.7+bpo-rt-amd64 is not configured yet.

dpkg: error processing package linux-headers-rt-amd64 (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 linux-image-6.9.7+bpo-rt-amd64
 linux-image-rt-amd64
 linux-headers-6.9.7+bpo-rt-amd64
 linux-headers-rt-amd64
E: Sub-process /usr/bin/dpkg returned an error code (1)
(exit=100)
```

</details>

----

<details>
<summary>file `/var/lib/dkms/v4l2loopback/0.12.7/build/make.log` contents (click to reveal)</summary>

```
DKMS make.log for v4l2loopback-0.12.7 for kernel 6.9.7+bpo-rt-amd64 (x86_64)
Mon Jul 22 03:48:09 PM CEST 2024
Building v4l2-loopback driver...
make -C /lib/modules/6.9.7+bpo-rt-amd64/build M=/var/lib/dkms/v4l2loopback/0.12.7/build modules
make[1]: Entering directory '/usr/src/linux-headers-6.9.7+bpo-rt-amd64'
  CC [M]  /var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.o
/var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.c: In function ‘vidioc_querycap’:
/var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.c:717:9: error: implicit declaration of function ‘strlcpy’; did you mean ‘strncpy’? [-Werror=implicit-function-declaration]
  717 |         strlcpy(cap->driver, "v4l2 loopback", sizeof(cap->driver));
      |         ^~~~~~~
      |         strncpy
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-6.9.7+bpo-common-rt/scripts/Makefile.build:249: /var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.9.7+bpo-common-rt/Makefile:1943: /var/lib/dkms/v4l2loopback/0.12.7/build] Error 2
make[1]: *** [/usr/src/linux-headers-6.9.7+bpo-common-rt/Makefile:252: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.9.7+bpo-rt-amd64'
make: *** [Makefile:43: v4l2loopback.ko] Error 2
```

</details>

----

umlaeute, 2024-07-22:
Sure.
You need `v4l2loopback>=0.13`, which is not available on Debian 12 (or backports).
The Debian12 package of `v4l2loopback` works nicely with the linux-kernel-headers package of Debian12.
But if you install a package from backports, then this is no longer guaranteed (and obviously fails in your case).

Abdull, 2024-07-23:
> Sure. You need `v4l2loopback>=0.13`, which is not available on Debian 12 (or backports). The Debian12 package of `v4l2loopback` works nicely with the linux-kernel-headers package of Debian12. But if [...]

Thank you for your response.

As I don't know what to do in such situations, I just removed `v4l2loopback` (i.e., `sudo apt remove v4l2loopback-dkms`).

Then after running `sudo apt upgrade`, Debian 12 was able to set up and install `[linux-image-6.9.7+bpo-rt-amd64](https://packages.debian.org/bookworm-backports/linux-image-6.9.7+bpo-rt-amd64)`, albeit without the `v4l2loopback-dkms`, and also rebuilding all other installed linux-image / kernels *without* v4l2loopback dynamic kernel module support.

umlaeute, 2024-07-24:
> As I don't know what to do in such situations

well, https://backports.debian.org/#introduction explicitly says:
> backports are provided on an as-is basis, with risk of incompatibilities with other components in Debian stable.