v4l2loopback/v4l2loopback #579: Possible Missing Dependency?

BeoHawk25, 2024-03-20:
# Missing Dependency
Long time user, first time issue poster. I'm always open to discussion, and I'm always trying to learn something. If I'm way off base, please help me understand why.

There may be a missing dependency for `debugedit` or 'base-devel'. 

In debugging my OBS Studio installation, I was swapping from the `v4l2loopback-dkms` package to the `v4l2loopback-dkms-git` [package on AUR](https://aur.archlinux.org/packages/v4l2loopback-dkms-git) (0.12.5.r232.g2fa9d6d-1). The build and installation it YAY failed, due to a missing binary (terminal output below). First installing the package for that binary allowed me to complete my installation of v4l2loopback.

I know that `debugedit` is included in `base-devel`, but neither is included as a dependency in the makefile (I saw the note about the kernal dependencies). I get that it may not be possible to add as a dependency, but it might be worth adding to the [Readme.md](https://github.com/umlaeute/v4l2loopback/blob/main/README.md) to specify that those are expected as part of the development environment for building the package?

### My Environment

  * `v4l2loopback` version:
```
sudo dmesg  | grep -i v4l2loopback
[ 4381.981837] v4l2loopback driver version 0.13.1 loaded
```

  * kernel version:
```
uname -a
Linux Kite 6.8.1-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Sat, 16 Mar 2024 17:15:23 +0000 x86_64 GNU/Linux
```

  * Distribution (+version):
```
lsb_release -a
LSB Version:    n/a
Distributor ID: Garuda
Description:    Garuda Linux
Release:        Soaring
Codename:       Spizaetus
```

### The Problem:

#### Steps to reproduce:

  1. Attempted an installation of the current release using YAY:
  ```
  yay -S v4l2loopback-dkms-git
AUR Explicit (1): v4l2loopback-dkms-git-0.12.5.r232.g2fa9d6d-1
Sync Make Dependency (1): help2man-1.49.3-1
:: PKGBUILD up to date, skipping download: v4l2loopback-dkms-git
1 v4l2loopback-dkms-git            (Build Files Exist)
==> Packages to cleanBuild?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> a
:: Deleting (1/1): /home/hawkins/.cache/yay/v4l2loopback-dkms-git
HEAD is now at 394829c Fixed source url.
Skipping repository src/v4l2loopback
Removing v4l2loopback/
1 v4l2loopback-dkms-git            (Build Files Exist)
==> Diffs to show?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> n
==> Making package: v4l2loopback-dkms-git 0.12.5.r232.g2fa9d6d-1 (Wed 20 Mar 2024 12:04:50 PM CDT)
==> Retrieving sources...
-> Cloning v4l2loopback git repo...
Cloning into bare repository '/home/hawkins/.cache/yay/v4l2loopback-dkms-git/v4l2loopback'...
remote: Enumerating objects: 4151, done.
remote: Counting objects: 100% (1040/1040), done.
remote: Compressing objects: 100% (475/475), done.
remote: Total 4151 (delta 623), reused 829 (delta 565), pack-reused 3111
Receiving objects: 100% (4151/4151), 1.33 MiB | 4.84 MiB/s, done.
Resolving deltas: 100% (2484/2484), done.
==> WARNING: Skipping verification of source file PGP signatures.
==> Validating source files with md5sums...
v4l2loopback ... Skipped
:: Remove make dependencies after install? [y/N] n
:: (1/1) Parsing SRCINFO: v4l2loopback-dkms-git
[... 360 lines cut ...]
libqaccessibilityclient-qt5 0.6.0-1
libxres 1.2.2-1
modemmanager-qt5 5.115.0-1
networkmanager-qt5 5.115.0-1
oxygen-sounds 6.0.2-3
purpose5 5.115.0-3
python-rx 3.2.0-4
qqc2-desktop-style5 5.115.0-1
wayland-protocols 1.33-1
webrtc-audio-processing 0.3.1-4
(7/9) Checking for .pacnew and .pacsave files...
.pac* files found:
/etc/passwd.pacnew
/etc/shells.pacnew
/etc/locale.gen.pacnew
/etc/pam.d/kde.pacnew
/etc/pacman.conf.pacnew
/etc/pacman.d/mirrorlist.pacnew
Please check and merge
(8/9) Performing snapper post snapshots for the following configurations...
==> root: 369
(9/9) Syncing all file systems...
```

4. OBS Studio is happy, my virtual camera is working again, and the world is once again filled with sunshine and kittens!

umlaeute, 2024-03-20:
i cannot reproduce this.
e.g. in a fresh Debian docker container i do:

```sh
$ apt-get update
$ apt-get install build-essential git linux-headers-amd64
$ git clone https://github.com/umlaeute/v4l2loopback.git
$ cd v4l2loopback

$ ls /lib/modules/
6.1.0-18-amd64
$ make KERNELRELEASE=6.1.0-18-amd64
Building v4l2-loopback driver...
make -C /lib/modules/6.1.0-18-amd64/build M=/root/v4l2loopback KCPPFLAGS="-DSNAPSHOT_VERSION='"0.13.1-1-gcb9e676"'" modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.0-18-amd64'
  CC [M]  /root/v4l2loopback/v4l2loopback.o
  MODPOST /root/v4l2loopback/Module.symvers
  CC [M]  /root/v4l2loopback/v4l2loopback.mod.o
  LD [M]  /root/v4l2loopback/v4l2loopback.ko
  BTF [M] /root/v4l2loopback/v4l2loopback.ko
Skipping BTF generation for /root/v4l2loopback/v4l2loopback.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-18-amd64'
make -C utils V4L2LOOPBACK_SNAPSHOT_VERSION=0.13.1-1-gcb9e676
make[1]: Entering directory '/root/v4l2loopback/utils'
cc  -I.. -DSNAPSHOT_VERSION='"0.13.1-1-gcb9e676"'  -c -o v4l2loopback-ctl.o v4l2loopback-ctl.c
cc   v4l2loopback-ctl.o   -o v4l2loopback-ctl
make[1]: Leaving directory '/root/v4l2loopback/utils'

$ debugedit
bash: debugedit: command not found
```

so i conclude that `v4l2loopback` does not depend on `debugedit`.

all in all it seems that this really is a problem with the packaging for your distribution (or with `yay` in general).

sorry to not be of any help

closed 2024-03-20 by umlaeute

umlaeute, 2024-03-20:
note that i am no expert in `arch` (and i never even heard of `Garuda`), but all the docs seem to clearly indicate that in order to use AUR packages, you *must* have `base-devel` installed.

quoting https://wiki.archlinux.org/title/Arch_User_Repository#Getting_started:

> ### Getting started
> 
> Users can search and download [PKGBUILDs](https://wiki.archlinux.org/title/PKGBUILD) from the [AUR Web Interface](https://aur.archlinux.org/). These PKGBUILDs can be built into installable packages using makepkg, then installed using pacman.
> 
> -  Ensure [base-devel](https://archlinux.org/packages/?name=base-devel) is [installed](https://wiki.archlinux.org/title/Install).

so to conclude: it is outside the scope of `v4l2loopback` to document generic requirements to get the package manager of an arbitrary distribution up and running.

umlaeute, 2024-03-20:
and just for the fun of it, trying to build the module in a minimal archlinux environment (`docker run --rm -ti archlinux`):

```sh
$ pacman -Sy
$ pacman -S gcc make git linux-headers
$ git clone https://github.com/umlaeute/v4l2loopback
$ cd v4l2loopback
$ ls /lib/modules
6.8.1-arch1-1
$ make KERNELRELEASE=6.8.1-arch1-1
Building v4l2-loopback driver...
make -C /lib/modules/6.8.1-arch1-1/build M=/v4l2loopback KCPPFLAGS="-DSNAPSHOT_VERSION='"0.13.1-1-gcb9e676"'" modules
make[1]: Entering directory '/usr/lib/modules/6.8.1-arch1-1/build'
  CC [M]  /v4l2loopback/v4l2loopback.o
  MODPOST /v4l2loopback/Module.symvers
  CC [M]  /v4l2loopback/v4l2loopback.mod.o
  LD [M]  /v4l2loopback/v4l2loopback.ko
  BTF [M] /v4l2loopback/v4l2loopback.ko
make[1]: Leaving directory '/usr/lib/modules/6.8.1-arch1-1/build'
make -C utils V4L2LOOPBACK_SNAPSHOT_VERSION=0.13.1-1-gcb9e676
make[1]: Entering directory '/v4l2loopback/utils'
cc  -I.. -DSNAPSHOT_VERSION='"0.13.1-1-gcb9e676"'  -c -o v4l2loopback-ctl.o v4l2loopback-ctl.c
cc   v4l2loopback-ctl.o   -o v4l2loopback-ctl
make[1]: Leaving directory '/v4l2loopback/utils'

$ debugedit
bash: debugedit: command not found

$
```

BeoHawk25, 2024-03-20:
Thanks for taking a look at it!

I'm certainly no Arch expert either, just slightly more adventurous than your average user. Totally get that it is beyond the scope to list another systems dependencies.

Like I mentioned I did figure out what I was missing and did get it running, so it's functional on my end now. It was just odd that this is the first time I've run into an issue looking for anything from the base-devel package.

Also it's very cool to see containers used in debugging this way! I'm familiar with them for deployment environments, very neat application of containers!

umlaeute, 2024-03-21:
> Totally get that it is beyond the scope to list another systems dependencies.

note that this isn't really about the dependencies of `v4l2loopback` (which doesn't use `debugedit` *at all*).
it's about the dependencies of your package manager itself.

in any case, i think this is covered by the last ellipsis in the [README](https://github.com/umlaeute/v4l2loopback#dependencies):
> In order to build (compile,...) anything, you must have a *working* build-environment (compiler, GNU make,**...**).