v4l2loopback/v4l2loopback #581: getNextFrame no data errno:0 Success on v4l2rtspserver log with verbose

javiluco24, 2024-03-21:
Step 2

About my environment

v4l2rtspserver Server Version: 0.3.8-5-ge2ad168 live555 version:2024.02.15

  * kernel version: 

  6.6.20+rpt-rpi-v8

  * Distribution (+version):

        Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm

Step 3: Describing my problem:

1. I have configured an USB 3.0 video HDI capture in my Raspberry PI 4 with 8GB RAM, 
2. Then I have plugged into HDMI port a video console, PS4.

executing the RTSP server as follow: v4l2rtspserver -c -I 0.0.0.0, sometimes (1 min, 2 min, randomly) get this on the log:

[NOTICE] /root/v4l2rtspserver/src/V4L2DeviceSource.cpp:214
	V4L2DeviceSource::getNextFrame no data errno:0 Success
[ERROR] /root/v4l2rtspserver/src/V4L2DeviceSource.cpp:117
	error:Success

immediately the streaming fails and capture freeze.

I use VLC as a client openning rtsp://SERVER_IP:8554/unicast

Could you provide me any solution?
Thanks in advance,
BR

umlaeute, 2024-03-21:
```sh
# apt-get install v4l2rtspserver
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package v4l2rtspserver
#
```

could i have access to your machine? ;-P

javiluco24, 2024-03-21:
Dear umlaeute,

Try "snap install v4l2rtspserver"; Version: 7c6ebae live555 version:2020.01.19

this is another version, but with the same result:

[NOTICE] /build/v4l2rtspserver/parts/v4lserver/src/src/V4L2DeviceSource.cpp:207
	V4L2DeviceSource::getNextFrame no data errno:0 Success
[ERROR] /build/v4l2rtspserver/parts/v4lserver/src/src/V4L2DeviceSource.cpp:111
	error:Success

on the other hand I installed v4l2rtspserver from source, Source code
(tar.gz)

both the same issue.

could i have access to your machine? yes, for sure, please tell me your IP add and let me create a user and accept your ssh request.

Thanks a lot,

BR

stephematician, 2024-10-18:
Try here: https://github.com/mpromonet/v4l2rtspserver