v4l2loopback/v4l2loopback #582: how create duplicate device0 and /dev/video1 who desn't have "Video Capture"?

LaurentDelaon, 2024-03-22:
Hi everybody,

I want to create and use multiple duplicate usb-camera. This hardware is on device0 ans device1   who desn't have "Video Capture" property.
Look at this link : [https://bugzilla.kernel.org/show_bug.cgi?id=199575](url).

Reason todo that is that you cannot have multiple  access on hardware device0 but possible with duplicate hardware.
I have a usb webcam on which define two devices : device0 and device1.
I do that and test read video4 and video 6 with vlc , it's works:
`//
sudo modprobe  v4l2loopback video_nr=4,5,6,7 exclusive_caps=1
//
ffmpeg -f video4linux2 -i /dev/video0 -codec copy -f v4l2 /dev/video4 -codec copy -f v4l2 /dev/video6`

Can also use :
`ffplay /dev/video4`

My problem come with an application who is perfectly working on video0 but unable vide video4 becaus ,as me,  it's can't found  the equivalant  duplicate of video1... say video5.

How can I solve this issue ?

umlaeute, 2024-03-22:
the loopback camera is **not** a USB-device. it doesn't attempt to be one.

also, if your application *requires* a metadata device, then i think it is broken.

i'm closing this as "out-of-scope" resp "won't fix". sorry

closed 2024-03-22 by umlaeute

LaurentDelaon, 2024-03-22:
Hi Umlaeute

As I write is it perfectly work when I connect the application  to video0 directly 
but don't  if I connect on video4 which is a copy of video0 thru v4l2loopback+ffmpeg
`ffmpeg -f video4linux2 -i /dev/video0 -codec copy -f v4l2 /dev/video4`

So how I can obtain via v4l2loopback the  device video1 which isn't 'videocapture'  ?

umlaeute, 2024-03-22:
as i said, you cannot.

your assumption is, that *your application* works only with `/dev/video0` *because* there is an accompanying `/dev/video1` metadata device.
if this assumption is correct, then i believe the application is broken (because there is no valid reason why there has to be a metadata device at all. i have seen plenty of real hardware capture devices that do not have a metadata device).
in this case, i'm closing this bug-report as "out-of-scope".

if your assumption is wrong, then there is no reason to duplicate the metadata device, and the problem is something else. in this case i'm closing this bug-report as "off-topic" (because then you should start a new bug report, that describes the actual problem)

umlaeute, 2024-03-22:
totally off-topic (and unrelated to your problem), but could you please explain why you chose to ignore the template-text of the bug-report? i try to get information [about the environment]((https://github.com/umlaeute/v4l2loopback/blob/main/.github/ISSUE_TEMPLATE/bug_report.md#step-2-describe-your-environment) early, because it is usually important.

i'm not asking you to provide this information right now, but wonder why you did not provide it in the first place.
i am asking because you are not alone, so i would like to improve my communication in this regard (to keep frustration levels low)

LaurentDelaon, 2024-03-22:
I am not english well readers speaker.
I haven't clearly understand pre writing.
Suggest simpler way :
Put in terminal commands and send  : sudo dmesg  | grep -i v4l2loopback....

```
sudo dmesg  | grep -i v4l2loopback
[sudo] Mot de passe de laurent : 
[ 2444.843240] v4l2loopback: loading out-of-tree module taints kernel.
[ 2444.843337] v4l2loopback: module verification failed: signature and/or required key missing - tainting kernel
[ 2444.844988] v4l2loopback driver version 0.12.7 (0.12.7-412-g850a2e3) loaded
[ 8244.940513] v4l2loopback driver version 0.12.7 (0.12.7-412-g850a2e3) loaded
laurent@laurent-NL40-50GU:~/programme/lazarus/pblireUsbCamFfmpeg$ uname -a
Linux laurent-NL40-50GU 5.4.0-150-generic #167~18.04.1-Ubuntu SMP Wed May 24 00:51:42 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
laurent@laurent-NL40-50GU:~/programme/lazarus/pblireUsbCamFfmpeg$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.6 LTS
Release:	18.04
Codename:	bionic
```