v4l2loopback/v4l2loopback #583: modprobe: ERROR: could not insert 'v4l2loopback': Exec format error

AmineMarnissi, 2024-03-28:
I installed it with these commands:
```
git clone https://github.com/umlaeute/v4l2loopback.git
cd v4l2loopback
sudo make
sudo make install
sudo depmod -a
```
When I run
`sudo modprobe v4l2loopback`
I get
modprobe: ERROR: could not insert 'v4l2loopback': Exec format error
From `sudo dmesg | grep -i v4l2loopback `
I get
[10448.341827] v4l2loopback: module PLT section(s) missing
[11603.729466] v4l2loopback: module PLT section(s) missing

From `uname -a `
I get
Linux EAC-5000 5.10.104-tegra #2 SMP PREEMPT Wed Nov 22 00:52:35 EST 2023 aarch64 aarch64 aarch64 GNU/Linux

From `lsb_release -a`
I get
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.6 LTS
Release:	20.04
Codename:	focal

moacirrf, 2024-03-28:
Hi, maybe it help
1. Try to identify if all dependencies are loaded:
    `modinfo v4l2loopback.ko`

will show something like

> depends:        videodev

2. load this module first
    `sudo modprobe videodev`

3. Try to load directly from the folder that you compile
    `sudo insmod v4l2loopback.ko`

3. check with dmesg command

umlaeute, 2024-04-02:
i strongly suspect this to be a duplicate of
- https://github.com/umlaeute/v4l2loopback/issues/313
- https://github.com/umlaeute/v4l2loopback/issues/260
- https://github.com/umlaeute/v4l2loopback/issues/238
- https://github.com/umlaeute/v4l2loopback/issues/208
- https://github.com/umlaeute/v4l2loopback/issues/147

(btw, you can search old issues (open or closed) to see whether your problem was already reported)

umlaeute, 2024-04-02:
after doing slightly more research (basically doing a websearch for ["PLT section(s) missing"](https://duckduckgo.com/?q=PLT+section(s)+missing&ia=web)), i found this Debian bug:
- [linux: No rule to make target 'scripts/module.lds' while building out-of-tree modules](https://bugs.debian.org/975571)

note that this is not really the same issue as the ones listed above.

however, it also confirms that the issue is not within `v4l2loopback`, but rather with the kernel packaging. the bug was discovered (and fixed) in 2020, so your "Ubuntu 20.04" might well be affected.

so i'm afraid this is out of scope for us.