v4l2loopback/v4l2loopback #587: [Bug]: Can't create second device

v4u6h4n, 2024-04-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Detail

```markdown
I can successfully create a virtual device for OBS studio, but I have been unable to create a second device. Not sure what the issue is as its literally the same command, with the card label and number changed. My current command looks like:

`sudo modprobe v4l2loopback exclusive_caps=1 video_nr=89,99 card_label="Virtual Logitech","OBS Virtual Camera"`
```

### v4l2loopback version

[   24.947632] v4l2loopback driver version 0.13.1 loaded

### kernel version

Linux desktop-0 6.8.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 17 Apr 2024 15:20:28 +0000 x86_64 GNU/Linux

### OS Version

Arch

### Which CPU are you using?

amd64/x86_64 ("64bit Intel")

umlaeute, 2024-04-26:
the `modprobe` command loads the kernel module.
as a side effect, this creates the loopback devices.

however, once you have loaded the kernel module, you cannot just load it again: it's already loaded.
(you can unload the module first, and then load it. but as a side effect, this will first remove all existing loopback devices).

prior to 0.13, you would have pass the maximum number of devices you expect during the lifetime of the running kernel at the time you load the module.

since 0.13, you can dynamically create devices with the `v4l2loopback-ctl` utility.

closed 2024-04-26 by umlaeute

v4u6h4n, 2024-04-26:
Hey @umlaeute 

The command I specified above is run at boot time, so it isn't an issue of the module needing to be unloaded. I didn't have `v4l2loopback-utils` installed until just a moment ago however, which resolved my issue. Is it needed for the command I specified to work correctly?

umlaeute, 2024-04-26:
Then I misunderstood your problem. 

Good that the issue had been resolved nevertheless. 

In any case: no, having `v4l2loopback-utils` installed is orthogonal to the module loading (and doesn't help with the `modprobe` functionality)

v4u6h4n, 2024-04-26:
Okay, well although the problem is solved, I may have encountered a bug with the module, as I haven't been able to have it create the device with the command I specified.

v4u6h4n, 2024-04-28:
Okay, I think there is some kind of bug with how v4l2loopback handles card_labels, or the documentation in the readme needs to be updated. If I add the following to `/etc/modprobe.d/v4l2loopback.conf`:

```
options v4l2loopback exclusive_caps=1 video_nr=50,51,99 card_label="webcam1,webcam2,OBS Virtual Camera"
```

I get the following devices on startup:

```
OUTPUT          CAPTURE         NAME
/dev/video50    /dev/video50    OBS Virtual Camera
/dev/video51    /dev/video51    webcam2
/dev/video99    /dev/video99    OBS Virtual Camera
```

The readme says the following, so I think I'm structuring correctly...

```
Note that the double quotes can only be used at the beginning and the end of the option's value, as opposed to when they are specified on the command line.
```

umlaeute, 2024-04-28:
when running your command:

```sh
rmmod v4l2loopback
modprobe v4l2loopback exclusive_caps=1 video_nr=89,99 card_label="Virtual Logitech","OBS Virtual Camera"
```

i get (as expected):
```sh
$ v4l2loopback-ctl list
OUTPUT     	CAPTURE      	NAME
/dev/video89 	/dev/video89 	Virtual Logitech
/dev/video99 	/dev/video99 	OBS Virtual Camera 
```

so i do not think there's a bug here.

the quoting is POSIX compliant, so unless you are using some very exotic shell to issue the command there shouldn't be any problem.
otoh, afaict you never said what you *actually* get when issuing your `modprobe` command.

v4u6h4n, 2024-04-30:
Hey @umlaeute 

**TLDR**
Apologies, I got confused about which thing was causing the issue, the issue is caused by the `/etc/modprobe.d/v4l2loopback.conf`, and not by the shell command as I originally indicated.

---

My apologies for the confusion, I haven't communicated clearly enough; I realised that my original issue report was in error. I tried to approaches to setting up the virtual devices, the first was the shell command (my original issue post), and the second was using `/etc/modprobe.d/v4l2loopback.conf`. I got confused about which one was causing the issue, the `/etc/modprobe.d/v4l2loopback.conf` is actually where the issue is coming from. So the issue is described correctly in [this post](https://github.com/umlaeute/v4l2loopback/issues/587#issuecomment-2081278600).

umlaeute, 2024-04-30:
np. i'm confused all the time.