v4l2loopback/v4l2loopback #596: [Bug]: BTF errors against kernel 6.10.5

potuz, 2024-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Detail

```markdown
Apologies to open an issue for support, but I have been trying to compile this module against 6.10.5 for a few days now and I can't modprobe it. The same procedure works fine against 6.5.1. I tried getting support in the Arch Linux forums without much success. 

$ uname -a
Linux forkchoice 6.10.5 #3 SMP PREEMPT_DYNAMIC Wed Aug 14 21:39:10 -03 2024 x86_64 GNU/Linux
$ make clean
...
$ make
...
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/heluani/Documents/code/video-drivers/v4l2loopback KCPPFLAGS="-DSNAPSHOT_VERSION='"0.13.2-1-g9318243"'" modules
make -C utils V4L2LOOPBACK_SNAPSHOT_VERSION=0.13.2-1-g9318243
make[1]: Entering directory '/home/potuz/Documents/code/video-drivers/v4l2loopback/utils'
cc  -I.. -DSNAPSHOT_VERSION='"0.13.2-1-g9318243"'  -c -o v4l2loopback-ctl.o v4l2loopback-ctl.c
make[1]: Entering directory '/home/potuz/Documents/code/kernel/linux-6.10.5'
cc   v4l2loopback-ctl.o   -o v4l2loopback-ctl
make[1]: Leaving directory '/home/potuz/Documents/code/video-drivers/v4l2loopback/utils'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 14.2.1 20240805
  You are using:           gcc (GCC) 14.2.1 20240910
  CC [M]  /home/potuz/Documents/code/video-drivers/v4l2loopback/v4l2loopback.o
  MODPOST /home/potuz/Documents/code/video-drivers/v4l2loopback/Module.symvers
  CC [M]  /home/potuz/Documents/code/video-drivers/v4l2loopback/v4l2loopback.mod.o
  LD [M]  /home/potuz/Documents/code/video-drivers/v4l2loopback/v4l2loopback.ko
  BTF [M] /home/potuz/Documents/code/video-drivers/v4l2loopback/v4l2loopback.ko
make[1]: Leaving directory '/home/potuz/Documents/code/kernel/linux-6.10.5'
$ sudo modprobe v4l2loopback exclusive_caps=1
modprobe: ERROR: could not insert 'v4l2loopback': Invalid argument
$ sudo dmesg | tail
Sep14 07:37] [Sat Sep 14 07:37:35 2024] BPF: 	32_NOTE_MASK type_id=13 bits_offset=8352
[  +0.000007] [Sat Sep 14 07:37:35 2024] BPF:
[  +0.000001] [Sat Sep 14 07:37:35 2024] BPF: Invalid name
[  +0.000001] [Sat Sep 14 07:37:35 2024] BPF:
[  +0.000002] [Sat Sep 14 07:37:35 2024] failed to validate module [v4l2loopback] BTF: -22
```
This is all the context I get on the dmesg error. 
```
$ grep BTF .config
CONFIG_DEBUG_INFO_BTF=y
CONFIG_PAHOLE_HAS_SPLIT_BTF=y
CONFIG_DEBUG_INFO_BTF_MODULES=y
# CONFIG_MODULE_ALLOW_BTF_MISMATCH is not set
CONFIG_PROBE_EVENTS_BTF_ARGS=y
```
I don't think I am forced to allow for BTFs mismatches right? I am happy to provide any requested information.
```

### v4l2loopback version

v0.13.2-1-g9318243

### kernel version

Linux forkchoice 6.10.5 #3 SMP PREEMPT_DYNAMIC Wed Aug 14 21:39:10 -03 2024 x86_64 GNU/Linux

### OS Version

Arch Linux, kernel manually compiled

### Which CPU are you using?

amd64/x86_64 ("64bit Intel")

umlaeute, 2024-09-12:
1. i think the `BTF` error is a red herring. probably getting a bit more context from `dmesg` could help debugging the issue
2. i do not see any installation of the newly compiled module (`make install`), so I guess you are actually trying to load an old/... version of the module that lingers in `/usr/lib/$(uname  -r)`. try using `modprobe ./v4l2loopback.ko` (or `insmod ./v4l2loopback.ko`) to force loading the just-bulit module (or run `make install` before). you might also need to run `depmod -a`)

potuz, 2024-09-13:
Apologies I just simply forgot to copy/paste the `sudo make install` step, depmod is unnecessary in my system, but the same error exists either way, and the same error happens with modprobing directly the just built file. I'll update the issue description with the install + depmod phases.

potuz, 2024-09-21:
I think this can be safely closed, I am finding BTFs errors on other modules now, this must be something on my local configuration, apologies for the noise.

closed 2024-09-21 by potuz