v4l2loopback/v4l2loopback #597: [Bug]: Write FPS configuration cannot be viewed

CoiicaulnxflleenHernandez6g3gxpvz, 2024-10-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Detail

```markdown
Version:v0.13.2
bug:The frame rate is written as 24 through $ echo '@24' | sudo tee /sys/devices/virtual/video4linux/video0/format. The previous version can be seen in /sys/devices/virtual/video4linux/video0/format. The current version (0.13.2) is empty, which makes it impossible to parse the frame rate through this file. It can be obtained by other means. Not sure if this problem is a bug or the storage location has been changed.
```

### v4l2loopback version

v0.13.2

### kernel version

```text
Linux version 5.10.120-tegra (root@JamesCheng-ub2004) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot 2020.08) 9.3.0, GNU ld (GNU Binutils) 2.33.1) #1 SMP PREEMPT Fri Nov 17 16:17:07 CST 2023
```

### OS Version

Ubuntu 20.04

### Which CPU are you using?

arm64 ("64bit ARM"; e.g. Apple Silicon,...)

umlaeute, 2024-10-16:
afaict, the problem is that now *reading* ` /sys/devices/virtual/video4linux/video0/format` only returns a format if all components (dimensions, colorspace, framerate) are fixed.
if there's no producer attached, then the format usually is not fixed at all, and you do not get a reading.

the question is: what would you expect as a result in such a case?

closed 2024-10-16 by CoiicaulnxflleenHernandez6g3gxpvz

umlaeute, 2024-10-16:
I think your question was quite valid.
(I just do not have a good answer what the output should really be)

But if the problem has been solved for you (or at least: clarified), all the better