v4l2loopback/v4l2loopback #603: Add minimal required CONFIG options

saloniamatteo, 2024-11-11:
Greetings,
after tinkering with Kernel options, since `v4l2loopback` was unable to build, I finally came up with a minimal list of required CONFIG options:

```
CONFIG_USB_VIDEO_CLASS=m
CONFIG_USB_VIDEO_CLASS_INPUT_EVDEV=y
CONFIG_V4L2_FWNODE=m
CONFIG_V4L_MEM2MEM_DRIVERS=y
CONFIG_V4L_PLATFORM_DRIVERS=y
CONFIG_V4L_TEST_DRIVERS=y
CONFIG_VIDEO=y
CONFIG_VIDEOBUF2_CORE=m
CONFIG_VIDEOBUF2_MEMOPS=m
CONFIG_VIDEOBUF2_V4L2=m
CONFIG_VIDEOBUF2_VMALLOC=m
CONFIG_VIDEO_DEV=m
CONFIG_VIDEO_MEM2MEM_DEINTERLACE=m
CONFIG_VIDEO_MUX=m
CONFIG_VIDEO_V4L2_I2C=y
```

(Perhaps even this list can be made smaller)

Please check if you have `CONFIG_TRIM_UNUSED_KSYMS` enabled: if it is, it WILL NOT allow you to compile v4l2loopback, unless you manually set the whitelist. Some (if not all) of the functions needed by v4l2loopback are:

```
__video_register_device()
ktime_get_ts64()
v4l2_ctrl_handler_free()
v4l2_ctrl_handler_init_class()
v4l2_ctrl_new_custom()
v4l2_ctrl_subscribe_event()
v4l2_device_register()
v4l2_event_queue()
v4l2_fill_pixfmt()
v4l2_fill_pixfmt_mp()
video_devdata()
video_device_alloc()
video_device_release()
video_ioctl2()
video_unregister_device()
vm_insert_page()
```

umlaeute, 2024-11-11:
> (Perhaps even this list can be made smaller)

on my system (linux-6.11.4, Debian kernel package), `CONFIG_VIDEO_MEM2MEM_DEINTERLACE` is not set, and `CONFIG_VIDEO_MUX` is not even present in the configuration (suggesting that there's no such option).

I don't think any of `CONFIG_VIDEO_V4L2_I2C`, `CONFIG_USB_VIDEO_CLASS` and `CONFIG_USB_VIDEO_CLASS_INPUT_EVDEV` are actually necessary (they seem to be hardware-specific), but haven't tested building a kernel without them.

umlaeute, 2024-11-11:
btw, thanks for checking this out.

i think the best place would be make this a Q&A in the [discussion](https://github.com/umlaeute/v4l2loopback/discussions/categories/q-a) section (and it probably should mention the kernel version this was tested against).

and yes, the rest of the config options look pretty sane to me :-)

saloniamatteo, 2024-11-13:
No problem.

Opened discussion #604, closing this issue.

closed 2024-11-13 by saloniamatteo

umlaeute, 2024-11-13:
thanks.
you probably should add your actual findings in a reply in #604 (so it can be marked as *answer*).
I could do that, but then the attribution is wrong.

saloniamatteo, 2024-11-13:
@umlaeute thanks, I'm now adding the replies and will continue to do further testing.