v4l2loopback/v4l2loopback #606: [Bug]: strange strings cripple the v4l2loopback-ctl man page

Vrihub, 2024-11-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Detail

```markdown
The man page for `v4l2loopback-ctl` contains many occurrences of strings like `==================` that disrupt readability.

I had a peek at the sources and it seems those strings belong to some intermediate markup language, but I don't think they should appear as such in the man page, so I guess something is wrong in the machinery that builds the page.
```

### v4l2loopback version

0.13.2

### kernel version

```text
6.11.5
```

### Which CPU are you using?

amd64/x86_64 ("64bit Intel")

umlaeute, 2024-11-26:
the manpage is generate from `v4l2loopback-ctl -h` with the help of `help2man`.

patches to improve the output of `v4l2loopback-ctrl -h` so that it *also* renders nicely in the generated manpage are welcome.

Vrihub, 2024-11-26:
Sorry I'm not proficient in C and help2man to contribute a patch.

Anyway here is a detailed description of the problem:
- The problem is in the section headings lines, e.g. "general commands" etc
- In the man page those lines appear over-indented by 2 levels, they should have 0 amount of indentation instead
- In the man page the "===...===" marker strings should be omitted

closed 2024-11-26 by umlaeute

Vrihub, 2024-11-27:
Looks very good now. Thank you very much!