v4l2loopback/v4l2loopback #607: [Bug]: Our producer is always getting the same (shared mem) buffer and so does the example producer

Misterke, 2024-12-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Detail

```markdown
We are using v4l2loopback to pass uncompressed video frames between two applications, but were noticing tearing and other "corruption" as if the buffer that the consumer received was actually being overwritten by the producer before the consumer queued it again.

We noticed the producer was every time getting the same buffer from DQBUF and the consumer then also always got that same buffer.

Then we tried the test applications "producer" and "consumer" that are provided with v4l2loopback and also see similar behavior:

~/v4l2loopback/tests$ ./producer -c -1
get format: OUTPUT:640x480:YUYV bytes/line=1280 sizeimage=614400 field=none
tried format: OUTPUT:640x480:YUYV bytes/line=1280 sizeimage=614400 field=none
got format: OUTPUT:640x480:YUYV bytes/line=1280 sizeimage=614400 field=none
set format: OUTPUT:640x480:YUYV bytes/line=1280 sizeimage=614400 field=none
finalizing format: OUTPUT:640x480:YUYV bytes/line=1280 sizeimage=614400 field=none
final format: OUTPUT:640x480:YUYV bytes/line=1280 sizeimage=614400 field=none
requested 4 buffers, got 2
requested buffer 0/4: buffer#0 @ 0x7ffe93838e50 OUTPUT bytesused=614400, length=614400 flags=0x00000002 field=none timestamp=6676956.317695 memory=MMAP (offset=0)
buffer#0 @0x7ba3b3167000 of 614400 bytes
requested buffer 1/4: buffer#1 @ 0x7ffe93838e50 OUTPUT bytesused=614400, length=614400 flags=0x00000002 field=none timestamp=6676956.317695 memory=MMAP (offset=614400)
buffer#1 @0x7ba3b30d1000 of 614400 bytes
MMAP init qbuf 0/2 (length=614400): buffer#0 @ 0x7ffe93839020 OUTPUT bytesused=614400, length=614400 flags=0x00000000 field=any timestamp=0.000000 memory=MMAP (offset=0)
MMAP init qbuf 1/2 (length=614400): buffer#1 @ 0x7ffe93839020 OUTPUT bytesused=614400, length=614400 flags=0x00000000 field=any timestamp=0.000000 memory=MMAP (offset=0)
MMAP    buffer#1 @ 0x7ffe93839000 OUTPUT bytesused=614400, length=614400 flags=0x00000001 field=none timestamp=0.000000 memory=MMAP (offset=614400)
MMAP    buffer#1 @ 0x7ffe93839000 OUTPUT bytesused=614400, length=614400 flags=0x00000001 field=none timestamp=0.000000 memory=MMAP (offset=614400)
MMAP    buffer#1 @ 0x7ffe93839000 OUTPUT bytesused=614400, length=614400 flags=0x00000001 field=none timestamp=0.000000 memory=MMAP (offset=614400)
MMAP    buffer#1 @ 0x7ffe93839000 OUTPUT bytesused=614400, length=614400 flags=0x00000001 field=none timestamp=0.000000 memory=MMAP (offset=614400)
MMAP    buffer#1 @ 0x7ffe93839000 OUTPUT bytesused=614400, length=614400 flags=0x00000001 field=none timestamp=0.000000 memory=MMAP (offset=614400)
MMAP    buffer#1 @ 0x7ffe93839000 OUTPUT bytesused=614400, length=614400 flags=0x00000001 field=none timestamp=0.000000 memory=MMAP (offset=614400)
MMAP    buffer#1 @ 0x7ffe93839000 OUTPUT bytesused=614400, length=614400 flags=0x00000001 field=none timestamp=0.000000 memory=MMAP (offset=614400)
MMAP    buffer#1 @ 0x7ffe93839000 OUTPUT bytesused=614400, length=614400 flags=0x00000001 field=none timestamp=0.000000 memory=MMAP (offset=614400)
MMAP    buffer#1 @ 0x7ffe93839000 OUTPUT bytesused=614400, length=614400 flags=0x00000001 field=none timestamp=0.000000 memory=MMAP (offset=614400)
MMAP    buffer#1 @ 0x7ffe93839000 OUTPUT bytesused=614400, length=614400 flags=0x00000001 field=none timestamp=0.000000 memory=MMAP (offset=614400)
```
Notice that also this producer is always getting buffer#1 despite having received 2 buffers at initialization.

So, is this a huge bug in v4l2loopback or are we totally missing how passing buffers through v4l2loopback is supposed to work?
```

### v4l2loopback version

v0.13.2-13-ge750af9

### kernel version

```text
Linux rd-int-nuc-1 6.8.0-45-generic #45-Ubuntu SMP PREEMPT_DYNAMIC Fri Aug 30 12:02:04 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
(but also seen on ARM architecture)
```

### OS Version

Ubuntu 24.04.1 LTS

### Which CPU are you using?

amd64/x86_64 ("64bit Intel"), arm64 ("64bit ARM"; e.g. Apple Silicon,...)

stephematician, 2024-12-12:
Can you provide a link to your source code? I think QBUF needs to be called on all buffers when streaming begins.

Misterke, 2024-12-12:
@stephematician , the source code is simply that of the producer and consumer in the tests folder of v4l2loopback itself.  If you run that producer, it already mentions that it is always getting the same buffer on every dqbuf.

stephematician, 2024-12-13:
Looks like this was introduced in 97dc86b. NFI if this was intended. v4l2loopback maintains an 'outbuffer list'; when dequeue occurs; it returns the _tail_ of that list, then moves the tail to ... the tail. It absolutely looks like a bug to me; and isn't linked to any 'issue' that I can find.

```c
		b = list_entry(dev->outbufs_list.prev, struct v4l2l_buffer,
			       list_head);
		list_move_tail(&b->list_head, &dev->outbufs_list);
```

stephematician, 2024-12-21:
Fairly sure this is a duplicate of #191 and #456.

commit 5092ea9a (2025-02-06): Revert 97dc86b and fix #607

Revert to FIFO management of dev->outbuf_list
closed 2025-02-06 by umlaeute