v4l2loopback/v4l2loopback #619: [Bug]: <title>v4l2loopback always creates non-capture device (caps 0x05200002) on Ubuntu 24.04.2

shayaShav, 2025-02-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Detail

```markdown
**First:**
* Similar bugs have been reported before, but I couldn’t find a working solution.
* Since my OS install is new (but not fresh), this report isn’t exactly a “bug” report. - it’s more of a request for help troubleshooting the issue. Thanks in advance and kudos for this awesome project!
----
<br>
I'm running **Ubuntu 24.04.2 LTS (6.8.0-53-generic)** with v4l2loopback `0.13.2-52-g77bf1e2` from GitHub.  
The Mesa version is `25.0~git2502130600.3a8abf~oibaf~n`.  
I have three physical "Insta360 Link 2C" USB cameras connected to the machine.

Whenever I remove and reload the module with a single device using `exclusive_caps=1`, I run:

$ sudo modprobe -r v4l2loopback
$ sudo modprobe v4l2loopback \
    devices=1 \
    video_nr=10 \
    card_label="USB Camera 1" \
    exclusive_caps=1

The created device always shows non-capture capabilities (Device Caps: 0x05200002):

$ sudo v4l2-ctl -d /dev/video10 --all
Driver Info:
        Driver name      : v4l2 loopback
        Card type        : USB Camera 1
        Bus info         : platform:v4l2loopback-010
        Driver version   : 6.8.12
        Capabilities     : 0x85200002
                Video Output
                Read/Write
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x05200002
                Video Output
                Read/Write
                Streaming
                Extended Pix Format
Priority: 2
Video output: 0 (loopback in)
Format Video Output:
        Width/Height      : 640/480
        Pixel Format      : 'BGR4' (32-bit BGRA/X 8-8-8-8)
        Field             : None
        Bytes per Line    : 2560
        Size Image        : 1228800
        Colorspace        : sRGB
        Transfer Function : Default (maps to sRGB)
        YCbCr/HSV Encoding: Default (maps to ITU-R 601)
        Quantization      : Default (maps to Full Range)
        Flags             : 
Streaming Parameters Video Output:
        Capabilities     : timeperframe
        Frames per second: 30.000 (30/1)
        Write buffers    : 2
User Controls
        keep_format 0x0098f900 (bool)   : default=0 value=0
        sustain_framerate 0x0098f901 (bool)   : default=0 value=0
        timeout 0x0098f902 (int)    : min=0 max=100000 step=1 default=0 value=0
        timeout_image_io 0x0098f903 (button) : value=0 flags=write-only, execute-on-write

-----
<br>

1. In my initial tests, everything worked as expected and the created devices had capture capabilities. I was able to stream video from FFmpeg into the new `/dev/video10` device, and Chrome recognized it as a USB camera.
2. After a few tests, the “non-capture” issue started.

**What I have tried and verified so far:**
1. The physical video devices are not busy when the issue occurs.
2. No errors appear in `dmesg` when the issue happens (only the v4l2-loopback driver version 0.13.2 (0.13.2-52-g77bf1e2) is loaded).
3. There are no custom rules applied in `/lib/udev/rules.d/`.
4. I discovered that when reloading with multiple devices (i.e., using `video_nr=10,12,14`) and `exclusive_caps=1`, the first device always fails to have capture capabilities, while the subsequent ones do. However, even though I can stream video into the new video device and see it on FFplay, Chrome does not recognize it as a camera.
5. I created a module options file (`/etc/modprobe.d/v4l2loopback.conf`) to load the module on boot:
   
   options v4l2loopback devices=4 video_nr=8,10,12,14 card_label="Test Cam,USB Camera 1,USB Camera 2,USB Camera 3" exclusive_caps=1
   
   Interestingly, after rebooting I observed that the created video devices had different names and numbers, and all had capture capabilities except for the first. However, even though I could stream video into the new video devices, Chrome still did not recognize them as cameras.

$ v4l2-ctl --list-devices
Insta360 Link 2C: Insta360 Link (platform:v4l2loopback-010):
        /dev/video10

Insta360 Link 2C: Insta360 Link (platform:v4l2loopback-011):
        /dev/video11

Insta360 Link 2C: Insta360 Link (platform:v4l2loopback-012):
        /dev/video12

USB Camera 3 (platform:v4l2loopback-014):
        /dev/video14

Insta360 Link 2C: Insta360 Link (usb-0000:00:14.0-2):
        /dev/video0
        /dev/video1
        /dev/media0

Insta360 Link 2C: Insta360 Link (usb-0000:00:14.0-8):
        /dev/video2
        /dev/video3
        /dev/media1

Insta360 Link 2C: Insta360 Link (usb-0000:00:14.0-9):
        /dev/video4
        /dev/video5
        /dev/media2

<br>
Any new investigation and debugging leads to detect what is the root of the problem would be greatly appreciated. Thanks!
```

### v4l2loopback version

0.13.2-52-g77bf1e2

### kernel version

```text
Linux ______ 6.8.0-53-generic #55-Ubuntu SMP PREEMPT_DYNAMIC Fri Jan 17 15:37:52 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux
```

### OS Version

Ubuntu/24.04.2 LTS (Noble Numbat)

### Which CPU are you using?

amd64/x86_64 ("64bit Intel")

stephematician, 2025-02-27:
This was fixed in 24cc1e82d; to be sure that this isn't a new bug - please confirm if the issue is resolved with the latest source.

stephematician, 2025-02-27:
Probably duplicate of #504

closed 2025-02-27 by umlaeute

shayaShav, 2025-03-05:
Thank you for the clear and detailed explanation @umlaeute ! 
It all makes perfect sense, and everything is working as expected now. 
I really appreciate it. Keep up the great work 🙌🏽

Vrihub, 2025-03-28:
I really think the clarifications you gave in 

https://github.com/v4l2loopback/v4l2loopback/issues/619#issuecomment-2688003269

should go into the documentation (man, readme, whatever), maybe in a more compact way, I can try to come up with a patch if you like. Or maybe for now you could add a link to it in the readme.

After not using the module for quite some time, today I upgraded it to 0.14.0, and I was about to file a bug because I thought I found a regression: `exclusive_caps=1` set at modprobe time didn't seem to apply to devices subsequently created via `v4l2loopback-ctl add`, i.e. I had to explicitly use `-x 1`, as you explained in your message (I don't rembember to ever have used `-x 1` in the past).

Has this always been the default behaviour? I seemed to remember that once you set `exclusive_caps=1` it would be inherited by all future created devices, but I might be wrong, since I haven't used the module for quite some time. Can you confirm this?

umlaeute, 2025-03-28:
> Has this always been the default behaviour?

yes

> i.e. I had to explicitly use `-x 1`, as you explained in your message (I don't remember to ever have used `-x 1` in the past).

i do not know about the last last time you used `v4l2loopback-add`, but it only grew device-adding capabilities with `0.13.x` (release 2024-03-19), and then you had to explicitly specify `-x 1`  if you wanted exclusive caps.

i do not remember exclusive caps being the default ever.

stephematician, 2025-03-28:
The _intended_ default behaviour was that `exclusive_caps` was _off_ when adding a device via _v4l2loopback-ctl_; this was restored in 2f121db - there was a logic error in the earlier code.