v4l2loopback/v4l2loopback #631: [Bug]: <title>Latest Pi OS kernel seems to have issue

rasheeddo, 2025-04-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Detail

```markdown
I have a Raspberry Pi 5 with v4l2loopback installed.
From `uname  -a` kernel version is `Linux raspberrypi 6.12.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux`

Create virtual device by,
`sudo modprobe v4l2loopback video_nr=40 exclusive_caps=1 card_label="Fake Device"`

Create fake device by pyfakewebcam

import time
import pyfakewebcam
import numpy as np

blue = np.zeros((720,1280,3), dtype=np.uint8)
blue[:,:,2] = 255

red = np.zeros((720,1280,3), dtype=np.uint8)
red[:,:,0] = 255

camera = pyfakewebcam.FakeWebcam('/dev/video40', 1280, 720)

while True:

    camera.schedule_frame(red)
    time.sleep(1/30.0)

    camera.schedule_frame(blue)
    time.sleep(1/30.0)

`ffplay /dev/video40` could visualize the red-blue screen from python script. But chromium browser couldn't recognize this fake device, neither Firefox.

But I have a snapshot image of Raspberry Pi 5 with older kernel,
`Linux raspberrypi 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux`

The same v4l2loopback and python script above is working fine, and chromium browser could recognize the fake device.

I think there is some update on RPi kernel that make v4l2loopback  couldn't work properly...
```

### v4l2loopback version

v0.14.0-22-g1195435 for new RPi kernel |  0.12.7 for old RPi kernel

### kernel version

```text
This kernel version,  v4l2loopback works  fine
`Linux raspberrypi 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux`

This kernel version v4l2loopback  doesn't work
`Linux raspberrypi 6.12.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux`
```

### OS Version

Raspberry Pi OS Bookworm  64-bit

### Which CPU are you using?

arm ("32bit ARM"; e.g. Raspberry Pi,...)

umlaeute, 2025-04-24:
what does `v4l2loopback-ctl query /dev/video40` show?

umlaeute, 2025-04-24:
> I think there is some update on RPi kernel that make v4l2loopback couldn't work properly...

well, you've updated both the kernel (`6.6.31+rpt-rpi-2712` -> `6.12.20+rpt-rpi-2712`) **and** v4l2loopback (`0.12.7` -> `v0.14.0-22-g1195435`), and maybe some other software in your stack.

many things could have gone wrong.

rasheeddo, 2025-04-24:
> in any case: what does v4l2loopback-ctl query /dev/video40 show?
On  0.12.7 (old kernel) I installed by `sudo apt install v4l2loopback-utils`, then the `v4l2loopback-ctl query /dev/video40` shows below
```
raspberry@raspberrypi:~ $ v4l2loopback-ctl query /dev/video40
Usage:
    /usr/bin/v4l2loopback-ctl [command] [args]

Commands:
     set-fps <FPS> /dev/videoX  Set device fps (if format is already
                                negotiated).

     set-timeout-image <IMAGE> /dev/videoX  Set timeout on device, and 
                                            optionally the placeholder picture,
                                            instead of default null frame.

     set-caps <CAPS> /dev/videoX  Fix format for the device (using GStreamer
                                  caps), so it will be maintained until reset,
                                  regardless of future readers/writers.

Options:
    FPS      frames per second, either as integer ('30') or fraction ('50/2')
    IMAGE    image file ("no-input.png"), or 'videotestsrc'
    CAPS     GStreamer caps ("video/x-raw, format=UYVY, width=640, height=480"),
             or 'any'
    /dev/videoX  the v4l2-loopback device file you want to act on
```
Maybe 0.12.7 doesn't have this `query` command?

On v0.14.0-22-g1195435 (new kernel), I installed by source from github, it doesn't seem to have `v4l2loopback-ctl`, did I miss something?

rasheeddo, 2025-04-24:
On both version, I  could use `ffplay /dev/video40` to see my test video, it works fine.
But the problem seems to be when I tried to check with [webcamtests.com](webcamtests.com), the browser couldn't recognize on new kernel, but the old one could.
Just trying to narrow it down to the issue, if you need any information, please let me know.

umlaeute, 2025-04-24:
> Maybe 0.12.7 doesn't have this query command?

yes.
it's only available since 0.13.0

i have currently no clue about the cause of the issue.

however, could you try:
- the new `v4l2loopback-0.14.x` with the old kernel?
- the old `v4l2loopback-0.12.7` with the new kernel?
   (if it doesn't compile, add a line `#define strlcpy strscpy` at the top of `v4l2loopback.c`)

rasheeddo, 2025-05-26:
> You could add `debug=2` to the module and look at the journal (system log) to see what happened when the browser tried to query the _loopback_ device (and, obviously, decided against using it).
> 
> Also - while I think that _pyfakewebcam_ is _not likely_ the problem - another way to test this would be to use _gstreamer_ with the pipeline:
> 
> gst-launch-1.0 videotestsrc ! "video/x-raw,height=480,width=640,format=YUY2" ! v4l2sink device=/dev/video40
> That should output a test pattern to the fake device which can be opened in _Chromium_ etc. You may or may not need to add `max_buffers=3` to the (_v4l2loopback_) module to convince _gstreamer_ to play nicely with _loopback_.

Hi, thanks for your comment.
I tested with RPi5 kernel 
```
Linux raspberrypi 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
```

The vl42loopback command as

```
sudo modprobe v4l2loopback video_nr=40 debug=2 max_buffers=3 exclusive_caps=1 card_label="Fake Device"
```

The Gstreamer command as

```
gst-launch-1.0 videotestsrc ! "video/x-raw,height=480,width=640,format=YUY2" ! videoconvert ! tee ! v4l2sink device=/dev/video40
```

Then I check on journalctl `journalctl -n 20`, then here is the log

```
May 26 06:11:28 raspberrypi kernel: /home/raspberry/v4l2loopback/v4l2loopback.c:1717[vidioc_reqbufs], pid(3390)
May 26 06:11:28 raspberrypi kernel: v4l2-loopback[2542], pid(3390):  allocate_buffers() size 1867776bytes = 622592bytes x 3buffers
May 26 06:11:28 raspberrypi kernel: v4l2-loopback[2548], pid(3390):  allocate_buffers() existing size=1867776bytes
May 26 06:11:28 raspberrypi kernel: v4l2-loopback[2552], pid(3390):  allocate_buffers() keep existing
May 26 06:11:28 raspberrypi kernel: /home/raspberry/v4l2loopback/v4l2loopback.c:1735[vidioc_reqbufs], pid(3390)
May 26 06:11:28 raspberrypi kernel: video40: VIDIOC_REQBUFS: count=2, type=vid-out, memory=mmap
May 26 06:11:28 raspberrypi kernel: video40: VIDIOC_QUERYBUF: 00:00:00.000000 index=0, type=vid-out, request_fd=0, flags=0x00002000, field=none, sequence=0, memory=mmap, bytesused=614400, offset/userptr=0x0, length=622592
May 26 06:11:28 raspberrypi kernel: timecode=00:00:00 type=0, flags=0x00000000, frames=0, userbits=0x00000000
May 26 06:11:28 raspberrypi kernel: video40: VIDIOC_QUERYBUF: 00:00:00.000000 index=1, type=vid-out, request_fd=0, flags=0x00002000, field=none, sequence=0, memory=mmap, bytesused=614400, offset/userptr=0x98000, length=622592
May 26 06:11:28 raspberrypi kernel: timecode=00:00:00 type=0, flags=0x00000000, frames=0, userbits=0x00000000
May 26 06:11:28 raspberrypi kernel: /home/raspberry/v4l2loopback/v4l2loopback.c:2204[v4l2_loopback_mmap], pid(3390)
May 26 06:11:28 raspberrypi kernel: /home/raspberry/v4l2loopback/v4l2loopback.c:2174[vm_open], pid(3390)
May 26 06:11:28 raspberrypi kernel: /home/raspberry/v4l2loopback/v4l2loopback.c:2204[v4l2_loopback_mmap], pid(3390)
May 26 06:11:28 raspberrypi kernel: /home/raspberry/v4l2loopback/v4l2loopback.c:2174[vm_open], pid(3390)
May 26 06:11:28 raspberrypi kernel: video40: VIDIOC_STREAMON: type=vid-out
May 26 06:13:53 raspberrypi systemd[1]: Starting fstrim.service - Discard unused blocks on filesystems from /etc/fstab...
May 26 06:13:55 raspberrypi fstrim[3430]: /: 44.7 GiB (47985774592 bytes) trimmed on /dev/mmcblk0p2
May 26 06:13:55 raspberrypi fstrim[3430]: /boot/firmware: 437 MiB (458278912 bytes) trimmed on /dev/mmcblk0p1
May 26 06:13:55 raspberrypi systemd[1]: fstrim.service: Deactivated successfully.
May 26 06:13:55 raspberrypi systemd[1]: Finished fstrim.service - Discard unused blocks on filesystems from /etc/fstab.
```

But still Chromium couldn't recognize the fake device.

Do you have any idea?

[image]

stephematician, 2025-05-26:
Open Chromium when the loopback device is being fed some frames, try to access the virtual camera (e.g. webcamtest), then provide a log.

The log you provided is for a successful use of the device, so it doesn't tell me much.

stephematician, 2025-05-26:
Also check that pipewire cameras are disabled in either Chromium or Firefox; if these apps are relying on pipewire, then the loopback device won't work. See https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/3659