v4l2loopback/v4l2loopback #632: [Bug]: V4l2sink (Gstreamer) is not working with HDMIRX video device

ErezAlster, 2025-04-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Detail

```markdown
I want to use gstreamer to analyze video and to source
For some reason I'm getting arguments error (22) when using v4l2-sink

I ran v4l2-compliance and got the following information
v4l2-compliance 1.26.1, 64 bits, 64-bit time_t

Compliance test for rk_hdmirx device /dev/video0:

Driver Info:
	Driver name      : rk_hdmirx
	Card type        : rk_hdmirx
	Bus info         : fdee0000.hdmirx-controller
	Driver version   : 6.1.99
	Capabilities     : 0x84201000
		Video Capture Multiplanar
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04201000
		Video Capture Multiplanar
		Streaming
		Extended Pix Format

Required ioctls:
		fail: v4l2-compliance.cpp(667): missing bus_info prefix ('fdee0000.hdmirx-controller')
	test VIDIOC_QUERYCAP: FAIL
		fail: v4l2-compliance.cpp(810): doioctl(node, ioc, nullptr) != ENOTTY
	test invalid ioctls: FAIL

Allow for multiple opens:
	test second /dev/video0 open: OK
		fail: v4l2-compliance.cpp(667): missing bus_info prefix ('fdee0000.hdmirx-controller')
	test VIDIOC_QUERYCAP: FAIL
	test VIDIOC_G/S_PRIORITY: OK
	test for unlimited opens: OK

Debug ioctls:
	test VIDIOC_DBG_G/S_REGISTER: OK (Not Supported)
	test VIDIOC_LOG_STATUS: OK

Input ioctls:
	test VIDIOC_G/S_TUNER/ENUM_FREQ_BANDS: OK (Not Supported)
	test VIDIOC_G/S_FREQUENCY: OK (Not Supported)
	test VIDIOC_S_HW_FREQ_SEEK: OK (Not Supported)
	test VIDIOC_ENUMAUDIO: OK (Not Supported)
	test VIDIOC_G/S/ENUMINPUT: OK
	test VIDIOC_G/S_AUDIO: OK (Not Supported)
	Inputs: 1 Audio Inputs: 0 Tuners: 0

Output ioctls:
	test VIDIOC_G/S_MODULATOR: OK (Not Supported)
	test VIDIOC_G/S_FREQUENCY: OK (Not Supported)
	test VIDIOC_ENUMAUDOUT: OK (Not Supported)
	test VIDIOC_G/S/ENUMOUTPUT: OK (Not Supported)
	test VIDIOC_G/S_AUDOUT: OK (Not Supported)
	Outputs: 0 Audio Outputs: 0 Modulators: 0

Input/Output configuration ioctls:
	test VIDIOC_ENUM/G/S/QUERY_STD: OK (Not Supported)
		fail: v4l2-test-io-config.cpp(210): field == V4L2_FIELD_NONE
		fail: v4l2-test-io-config.cpp(386): Timings check failed for input 0.
	test VIDIOC_ENUM/G/S/QUERY_DV_TIMINGS: FAIL
	test VIDIOC_DV_TIMINGS_CAP: OK
		fail: v4l2-test-io-config.cpp(571): ret != EINVAL
		fail: v4l2-test-io-config.cpp(625): EDID check failed for input 0.
	test VIDIOC_G/S_EDID: FAIL

Control ioctls (Input 0):
	test VIDIOC_QUERY_EXT_CTRL/QUERYMENU: OK
	test VIDIOC_QUERYCTRL: OK
	test VIDIOC_G/S_CTRL: OK
	test VIDIOC_G/S/TRY_EXT_CTRLS: OK
	test VIDIOC_(UN)SUBSCRIBE_EVENT/DQEVENT: OK
	test VIDIOC_G/S_JPEGCOMP: OK (Not Supported)
	Standard Controls: 4 Private Controls: 2

Format ioctls (Input 0):
	test VIDIOC_ENUM_FMT/FRAMESIZES/FRAMEINTERVALS: OK
	test VIDIOC_G/S_PARM: OK
	test VIDIOC_G_FBUF: OK (Not Supported)
		fail: v4l2-test-formats.cpp(374): ycbcr_enc >= 0xff
		fail: v4l2-test-formats.cpp(491): testColorspace(!node->is_io_mc, pix_mp.pixelformat, pix_mp.colorspace, pix_mp.ycbcr_enc, pix_mp.quantization)
	test VIDIOC_G_FMT: FAIL
		fail: v4l2-test-formats.cpp(744): Video Capture Multiplanar: TRY_FMT(G_FMT) != G_FMT
		warn: v4l2-test-formats.cpp(794): TRY_FMT cannot handle an invalid pixelformat.
		warn: v4l2-test-formats.cpp(795): This may or may not be a problem. For more information see:
		warn: v4l2-test-formats.cpp(796): http://www.mail-archive.com/linux-media@vger.kernel.org/msg56550.html
	test VIDIOC_TRY_FMT: FAIL
		warn: v4l2-test-formats.cpp(1157): S_FMT cannot handle an invalid pixelformat.
		warn: v4l2-test-formats.cpp(1158): This may or may not be a problem. For more information see:
		warn: v4l2-test-formats.cpp(1159): http://www.mail-archive.com/linux-media@vger.kernel.org/msg56550.html
		fail: v4l2-test-formats.cpp(1185): Video Capture Multiplanar: S_FMT(G_FMT) != G_FMT
	test VIDIOC_S_FMT: FAIL
	test VIDIOC_G_SLICED_VBI_CAP: OK (Not Supported)
	test Cropping: OK (Not Supported)
	test Composing: OK (Not Supported)
	test Scaling: OK (Not Supported)

Codec ioctls (Input 0):
	test VIDIOC_(TRY_)ENCODER_CMD: OK (Not Supported)
	test VIDIOC_G_ENC_INDEX: OK (Not Supported)
	test VIDIOC_(TRY_)DECODER_CMD: OK (Not Supported)

Buffer ioctls (Input 0):
		fail: v4l2-test-buffers.cpp(517): node->g_dv_timings(timings)
		fail: v4l2-test-buffers.cpp(642): testCanSetSameTimings(node)
	test VIDIOC_REQBUFS/CREATE_BUFS/QUERYBUF: FAIL
	test CREATE_BUFS maximum buffers: OK
	test VIDIOC_EXPBUF: OK
	test Requests: OK (Not Supported)

Total for rk_hdmirx device /dev/video0: 46, Succeeded: 37, Failed: 9, Warnings: 6

The main issue is that I'm able to use loopback device with ffmpeg (see example):
#!/bin/bash
sudo modprobe -r v4l2loopback
sudo modprobe v4l2loopback -f devices=2 buffer=2 video_nr=12,13
ffmpeg -f v4l2 -input_format bgr24 -video_size 1920x1080 -i /dev/video0 -pix_fmt bgr24 -f v4l2  /dev/video12 -f v4l2  /dev/video13

Any idea what should I do?
```

### v4l2loopback version

Latest from main

### kernel version

```text
Linux rock-5b-plus 6.1.99-vendor-rk35xx #1 SMP Mon Feb 17 13:23:33 UTC 2025 aarch64 aarch64 aarch64 GNU/Linux
```

### OS Version

Armbian 

### Which CPU are you using?

arm64 ("64bit ARM"; e.g. Apple Silicon,...)

umlaeute, 2025-04-29:
please provide the exact version of `v4l2loopback`,

> Latest from main

is *very* vague and means something different every other day.

---

also, while your report includes some information about the input source, there is no info about the actual error you are getting.

please any errors you are getting.

also, you're title says "gstreamer", but your script  uses "ffmpeg".

stephematician, 2025-05-19:
We need:

1.  The gstreamer command (or similar) that you were trying, and what the output was (see [Basic writing and formatting syntax - GitHub Docs](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code) regarding putting output in verbatim format)
2.  The commit or tag for _v4l2loopback_ that you were using.

_gstreamer_ can be rather strict about the data that it handles, so it is possible that it will fail with a device that isn't passing _v4l2-compliance_.

Instead of sinking to a `v4l2sink` (if that's what you are using), try sinking to `autovideosink`. If that does work - then we can consider whether this is actually a problem with _loopback_.