v4l2loopback/v4l2loopback #635: [Bug]: Video passthrough fails

Martmists-GH, 2025-05-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Detail

I tried to make a new virtual device that allows multiple applications to read from the video device:
```
$ sudo modprobe v4l2loopback video_nr=10 card_label="webcam-access" exclusive_caps=1
$ gst-launch-1.0 -v v4l2src device="/dev/video0" ! v4l2sink device=/dev/video10
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0: crop-bounds = < (int)0, (int)0, (int)1920, (int)1080 >
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = video/x-raw(memory:DMABuf), format=(string)DMA_DRM, drm-format=(string)NV12, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)120/1, interlace-mode=(string)progressive, colorimetry=(string)bt709
/GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0.GstPad:sink: caps = video/x-raw(memory:DMABuf), format=(string)DMA_DRM, drm-format=(string)NV12, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)120/1, interlace-mode=(string)progressive, colorimetry=(string)bt709

(gst-launch-1.0:91494): GStreamer-Video-CRITICAL **: 17:41:41.621: gst_video_frame_map_id: assertion 'info->finfo->format == meta->format' failed
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
../gstreamer/subprojects/gstreamer/libs/gst/base/gstbasesrc.c(3187): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason error (-5)
Execution ended after 0:00:00.442272736
Setting pipeline to NULL ...
Freeing pipeline ...
```
I have verified with lsof that no other program is currently using the video device.

### v4l2loopback version

extra/v4l2loopback-dkms 0.14.0-1

### kernel version

```text
Linux proton 6.14.4-zen1-2-zen #1 ZEN SMP PREEMPT_DYNAMIC Tue, 29 Apr 2025 09:23:22 +0000 x86_64 GNU/Linux
```

### OS Version

Arch Linux

### Which CPU are you using?

amd64/x86_64 ("64bit Intel")

Martmists-GH, 2025-05-04:
Using ffmpeg seems to work fine for *some* devices, but not others?

```
$ ffmpeg -re -i /dev/video2 -f v4l2 -codec:v rawvideo -pix_fmt yuv420p /dev/video10  # terminal 1
$ ffplay -fflags nobuffer /dev/video-webcam  # terminal 2
[works fine]
``` 

```
$ ffmpeg -re -i /dev/video0 -f v4l2 -codec:v rawvideo -pix_fmt yuv420p /dev/video10  # terminal 1
[video4linux2,v4l2 @ 0x64593a356500] Dequeued v4l2 buffer contains corrupted data (0 bytes).
    Last message repeated 30 times
$ ffplay -fflags nobuffer /dev/video-webcam  # terminal 2
[video4linux2,v4l2 @ 0x75d590000c80] Not a video capture device.
/dev/video-webcam: No such device
```

But also, trying to use the video devices from another tool such as OBS at the same time results in:
```
20:11:01.392: v4l2-input: Start capture from /dev/video-webcam
20:11:01.393: v4l2-input: Input: 0
20:11:01.393: v4l2-input: Unable to set format
20:11:01.393: v4l2-input: Initialization failed, errno: Device or resource busy
```

stephematician, 2025-05-19:
```console
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = video/x-raw(memory:DMABuf), format=(string)DMA_DRM, drm-format=(string)NV12, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)120/1, interlace-mode=(string)progressive, colorimetry=(string)bt709
/GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0.GstPad:sink: caps = video/x-raw(memory:DMABuf), format=(string)DMA_DRM, drm-format=(string)NV12, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)120/1, interlace-mode=(string)progressive, colorimetry=(string)bt709
```

These lines show that _gstreamer_ is trying to use DMA buffers; which _v4l2loopback_ does not support; at least - that's how I am reading the output.

1.  See what happens if you try `autovideosink` instead of `v4l2sink device=/dev/video10`; this _will not_ use the _loopback_ device, but it would be a good 'dry run' for _gstreamer_ and each source device you are using (`/dev/video0`, `/dev/video2` and `/dev/video-webcam`).
2.  Another thing to try is `autovideoconvert ! v4l2sink device=/dev/video10`, i.e. insert an [autovideoconvert](https://gstreamer.freedesktop.org/documentation/autodetect/autovideoconvert.html) from GStreamer "bad" plugins before the sink.
3.  You could also try to force memory mapped IO mode, e.g "v4l2sink device=/dev/video10 io-mode=mmap"; potentially on the source(s), too.

It may help us if you provide some brief details about what `/dev/video0`, `/dev/video2` and `/dev/video-webcam` are and what pixel formats (or buffers) you expect from them.

Martmists-GH, 2025-05-19:
1. It displays the feed from /dev/video0 as a window
2. 
```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0: crop-bounds = < (int)0, (int)0, (int)1920, (int)1080 >
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = video/x-raw(memory:DMABuf), format=(string)DMA_DRM, drm-format=(string)NV12, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)120/1, interlace-mode=(string)progressive, colorimetry=(string)bt709
Got context from element 'glcolorconvertelement3': gst.gl.GLDisplay=context, gst.gl.GLDisplay=(GstGLDisplay)"\(GstGLDisplayWayland\)\ gldisplaywayland0";
/GstPipeline:pipeline0/GstAutoVideoConvert:autovideoconvert0/GstBin:autovideoconvert-gluploadglcolorconvertglcolorscaleglcolorconvert.GstGhostPad:sink.GstProxyPad:proxypad9: caps = video/x-raw(memory:DMABuf), format=(string)DMA_DRM, drm-format=(string)NV12, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)120/1, interlace-mode=(string)progressive, colorimetry=(string)bt709
/GstPipeline:pipeline0/GstAutoVideoConvert:autovideoconvert0/GstBin:autovideoconvert-gluploadglcolorconvertglcolorscaleglcolorconvert.GstGhostPad:sink: caps = video/x-raw(memory:DMABuf), format=(string)DMA_DRM, drm-format=(string)NV12, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)120/1, interlace-mode=(string)progressive, colorimetry=(string)bt709
/GstPipeline:pipeline0/GstAutoVideoConvert:autovideoconvert0.GstPad:sink: caps = video/x-raw(memory:DMABuf), format=(string)DMA_DRM, drm-format=(string)NV12, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)120/1, interlace-mode=(string)progressive, colorimetry=(string)bt709
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0: crop-bounds = < (int)0, (int)0, (int)1920, (int)1080 >
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
../gstreamer/subprojects/gstreamer/libs/gst/base/gstbasesrc.c(3187): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.114684524
Setting pipeline to NULL ...
Freeing pipeline ...
```
3. That seems to run fine, however it still limits me to one client at a time on /dev/video10; Any additional clients will report `/dev/video10: Device or resource busy` despite creating the loopback with `exclusive_caps=1`

umlaeute, 2025-05-19:
>  That seems to run fine, however it still limits me to one client at a time on /dev/video10; Any additional clients will report `/dev/video10: Device or resource busy`

that is as expected: v4l2 does not allow more than one consumer to read a given video device (older versions of `v4l2loopback` ignored this, but this has now been *fixed*)

> despite creating the loopback with `exclusive_caps=1`

that's unrelated to the number of concurrent consumers; `exclusive_caps` **only** sets the behaviour to not announce CAPTURE and OUTPUT capabilities simultaneously (which is required by some clients).
regardless of this setting, you cannot have multiple consumers retrieve images from the device in parallel.

Martmists-GH, 2025-05-19:
> that is as expected: v4l2 does not allow more than one consumer to read a given video device (older versions of v4l2loopback ignored this, but this has now been fixed)

Which version should I downgrade to in that case? My setup doesn't lend itself to multiple loopback devices well, and I don't know how to make a virtual file that opens to a different internal loopback device created on-demand whenever a new client appears.

umlaeute, 2025-05-19:
`0.13.x` probably allows you to do this.

but *really*, this was **buggy** behaviour, you should adjust your setup to allow for the new (v4l2-compliant) behaviour.

stephematician, 2025-05-19:
If `io-mode=mmap` solved the problem, then this doesn't seem like an issue within _v4l2loopback_, this seems like an issue with _gstreamer_ and the devices you are using (which I asked for information about, but you seem to have missed that request).

As umlaeute said: concurrent readers are not supported by the V4L2 API (this is outside of _v4l2loopback_'s control); use `tee` from _gstreamer_ to duplicate the output to multiple devices instead.

closed 2025-05-27 by umlaeute