v4l2loopback/v4l2loopback #643: [Bug]: doesn't build on dkms linux-mainline 6.16-rc5

ManuLinares, 2025-07-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Detail

Installed linux-mainline 6.16rc5-1
dkms install failed

build/make.log
```
DKMS (dkms-3.2.1) make.log for v4l2loopback/0.15.0 for kernel 6.16.0-rc5-1-mainline (x86_64)
mar 08 jul 2025 18:32:20 -03

Building module(s)
# command: make -j16 KERNELRELEASE=6.16.0-rc5-1-mainline KERNEL_DIR=/usr/lib/modules/6.16.0-rc5-1-mainline/build v4l2loopback
grep: /etc/dkms/framework.conf.d/*.conf: No such file or directory

++++++ To sign the  module, you must set KBUILD_SIGN_KEY/KBUILD_SIGN_CERT to point to the signing key/certificate!
++++++ For your convenience, we try to read these variables as 'mok_signing_key' resp. 'mok_certificate' from /etc/dkms/framework.conf /etc/dkms/framework.conf.d/*.conf

++++++ If your certificate requires a password, pass it via the KBUILD_SIGN_PIN env-var!
grep: /etc/dkms/framework.conf.d/*.conf: No such file or directory
++++++ E.g. using 'export KBUILD_SIGN_PIN; read -s -p "Passphrase for signing key : " KBUILD_SIGN_PIN; sudo --preserve-env=KBUILD_SIGN_PIN make sign'

Building v4l2-loopback driver...
make -C /usr/lib/modules/6.16.0-rc5-1-mainline/build M=/var/lib/dkms/v4l2loopback/0.15.0/build KCPPFLAGS="" modules
make[1]: Entering directory '/usr/lib/modules/6.16.0-rc5-1-mainline/build'
make[2]: Entering directory '/var/lib/dkms/v4l2loopback/0.15.0/build'
  CC [M]  v4l2loopback.o
v4l2loopback.c: In function ‘v4l2_loopback_add’:
v4l2loopback.c:2904:9: error: implicit declaration of function ‘setup_timer’ [-Wimplicit-function-declaration]
 2904 |         setup_timer(&dev->sustain_timer, sustain_timer_clb, nr);
      |         ^~~~~~~~~~~
make[4]: *** [/usr/lib/modules/6.16.0-rc5-1-mainline/build/scripts/Makefile.build:287: v4l2loopback.o] Error 1
make[3]: *** [/usr/lib/modules/6.16.0-rc5-1-mainline/build/Makefile:2003: .] Error 2
make[2]: *** [/usr/lib/modules/6.16.0-rc5-1-mainline/build/Makefile:248: __sub-make] Error 2
make[2]: Leaving directory '/var/lib/dkms/v4l2loopback/0.15.0/build'
make[1]: *** [Makefile:248: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.16.0-rc5-1-mainline/build'
make: *** [Makefile:53: v4l2loopback.ko] Error 2

# exit code: 2
# elapsed time: 00:00:01
----------------------------------------------------------------

```

### v4l2loopback version

v4l2loopback-dkms 0.15.0-1

### kernel version

```text
Linux machine 6.16.0-rc5-1-mainline #1 SMP PREEMPT_DYNAMIC Sun, 06 Jul 2025 23:55:30 +0000 x86_64 GNU/Linux
```

### OS Version

Archlinux

### Which CPU are you using?

amd64/x86_64 ("64bit Intel")

umlaeute, 2025-07-12:
apparently our CI is building against `6.16.0-0.rc4.38.fc43.x86_64` (on Fedora) and [succeeds](https://raw.githubusercontent.com/wiki/v4l2loopback/v4l2loopback/build-state.svg)

i've also tried to reproduce this on Arch using today's [linux-mainline-headers](https://aur.archlinux.org/packages/linux-mainline-headers) (6.16.0-rc5-1-mainline) within a container, and could compile the module without problems.

all these tests were conducted with current `head` (0e32e59ad802a5301179c52d2c8fb5c0a5edc9d9)

so i think that there is inherent problem with the `v4l2loopback` module, and I would need step-by-step instructions on how to reproduce this in a container, or otherwise (on my Debian system).

(sidenote: given that 196ae48c0f08dfcdcad0aded4917cf8a303bf237 fixes *the other* problem with linux-6.16, i probably should do a bugfix release anyhow)

ManuLinares, 2025-07-12:
I can confirm, running the git version it compiles fine against 6.16.0-rc5-1-mainline

I Should close this bug on a new release.

xrn, 2025-07-13:
Fedora 43 - Linux fedora 6.16.0-0.rc5.250712g379f604cc3dc.50.fc43.x86_64 - I have the same issue as above

```
v4l2loopback.c:2904:9: error: implicit declaration of function ‘setup_timer’ [-Wimplicit-function-declaration]
```

Kenny-Dave, 2025-07-15:
Fedora 42 KDE 6.15.5-200.fc42.x86_64 I have the same issue I believe. Is there a workaround? I'm heavily reliant on this program for my work. Is there a kernel known to work that I could roll back to? 

```
DKMS (dkms-3.2.1) make.log for v4l2loopback/git for kernel 6.15.5-200.fc42.x86_64 (x86_64)
Tue 15 Jul 06:57:01 BST 2025

Building module(s)
# command: make -j12 KERNELRELEASE=6.15.5-200.fc42.x86_64 KERNEL_DIR=/lib/modules/6.15.5-200.fc42.x86_64/build v4l2loopback

++++++ To sign the  module, you must set KBUILD_SIGN_KEY/KBUILD_SIGN_CERT to point to the signing key/certificate!
++++++ For your convenience, we try to read these variables as 'mok_signing_key' resp. 'mok_certificate' from /etc/dkms/framework.conf

++++++ If your certificate requires a password, pass it via the KBUILD_SIGN_PIN env-var!
++++++ E.g. using 'export KBUILD_SIGN_PIN; read -s -p "Passphrase for signing key : " KBUILD_SIGN_PIN; sudo --preserve-env=KBUILD_SIGN_PIN make sign'

Building v4l2-loopback driver...
make -C /lib/modules/6.15.5-200.fc42.x86_64/build M=/var/lib/dkms/v4l2loopback/git/build KCPPFLAGS="-DSNAPSHOT_VERSION='"0.13.2-5-g581eb3b"'" modules
make[1]: Entering directory '/usr/src/kernels/6.15.5-200.fc42.x86_64'
make[2]: Entering directory '/var/lib/dkms/v4l2loopback/git/build'
  CC [M]  v4l2loopback.o
v4l2loopback.c: In function ‘buffer_written’:
v4l2loopback.c:1685:9: error: implicit declaration of function ‘del_timer_sync’ [-Wimplicit-function-declaration]
 1685 |         del_timer_sync(&dev->sustain_timer);
      |         ^~~~~~~~~~~~~~
make[4]: *** [/usr/src/kernels/6.15.5-200.fc42.x86_64/scripts/Makefile.build:203: v4l2loopback.o] Error 1
make[3]: *** [/usr/src/kernels/6.15.5-200.fc42.x86_64/Makefile:2043: .] Error 2
make[2]: *** [/usr/src/kernels/6.15.5-200.fc42.x86_64/Makefile:260: __sub-make] Error 2
make[2]: Leaving directory '/var/lib/dkms/v4l2loopback/git/build'
make[1]: *** [Makefile:260: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.15.5-200.fc42.x86_64'
make: *** [Makefile:53: v4l2loopback.ko] Error 2

# exit code: 2
# elapsed time: 00:00:01
----------------------------------------------------------------
```

ony, 2025-08-03:
6.16 got released and now available on NixOs where this module fails same way as in original report.

I guess that `defined` checks only for existence of macro wrapper. But if there is actual function, it will fail

https://github.com/v4l2loopback/v4l2loopback/blob/7546a42a66d856933a75c2b0896cb735e3ca00c0/v4l2loopback.c#L48-L50

Not sure it is reliable to test that we can call `timer_setup`.

Can check with
```zsh
cpp /dev/stdin <<< $'int abc();\n#if defined(abc)\nDefined function\n#endif'
```
```zsh
cpp /dev/stdin <<< $'int _abc();\n#define abc _abc\n#if defined(abc)\nDefined macro\n#endif'
```

Weird is that [latest 6.16.x](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/include/linux/timer.h?h=v6.16#n120) do use macro wrapper and sources I'm building against do have that too. So maybe that's ok.

But what I see is missing is `#include <linux/timer.h>` (notice it is `timer.h` instead of `time.h`).   
I guess in some versions of code there is implicit inclusion of it.